atomic14/diy-esp32-epub-reader #2: Possibility to use SPIFFs as MicroSD reader alternative

martinberlin, 2021-09-30:
I will get a new SD card of 2GB since I've only one and it's on the 3D-printer. 
But I was thinking maybe you can add also the option to use SPIFFS file system on the ESP32. I guess there will be space for a few books there and it makes the whole thing more portable since you don't need to carry around that monstrosity of hanging SPI cables ;)
Just a suggestion since I think only thing that changes is initialization, then it's just another File System to be read.
Then you can have a define constant of FS_TYPE and select between 2 systems.

cgreening, 2021-09-30:
Definitely possible - I've got generic code that will switch between them on another project - https://github.com/atomic14/esp32_sdcard_audio/tree/main/idf-wav-sdcard

I'll have a look at pulling it in.

cgreening, 2021-09-30:
I've added a quick test to this branch that should support SPIFFS - `feature/spiffs-support`

You'll need to upload the filesystem from platformio - I've included a sample book that works.

There's a bit of work to do on the image handling side of things, but give it a go and see if it works :)

martinberlin, 2021-09-30:
Awesome. I will try it out as soon as I can get some free time.

cgreening, 2021-09-30:
No rush :)

cgreening, 2021-09-30:
I've updated the code so that it doesn't extract the images to file - this makes SPIFFS pretty usable - you could probably fit a reasonable number of books on it (5-10).

closed 2021-09-30 by cgreening