atomic14/diy-esp32-epub-reader #5: Change the buttons to use interrupts to update the UI

cgreening, 2021-10-01:
At the moment we're polling the buttons waiting for someone to click them - it would be nicer if this was driven by interrupts.

martinberlin, 2021-10-03:
Nice idea. As a reference I also adapted a component to read the L58 touch of this epaper, it also uses events (callback functions) you can find it here:
https://github.com/martinberlin/FT6X36-IDF

and also some implementation examples here:
https://github.com/martinberlin/cale-idf/tree/master/main/demos/parallel

cgreening, 2021-10-03:
I'll take a look - I didn't support touch initially as I was using the SD Card which is on the same pins as the touch controller and there's no way to coordinate the two.

But when using SPIFFS the touch controller is available. Waking from deep sleep might still need to be done from the buttons as the touch controller is switched off when the EPD is powered down.

martinberlin, 2021-10-07:
One important to reminder is that I've read in [GPIO & RTC documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html#_CPPv416gpio_intr_enable10gpio_num_t) the following:

>Please do not use the interrupt of GPIO36 and GPIO39 (Used by DOWN button) when using ADC or Wi-Fi with sleep mode enabled. Please refer to the comments of adc1_get_raw. Please refer to section 3.11 of ‘ECO_and_Workarounds_for_Bugs_in_ESP32’ for the description of this issue. As a workaround, call adc_power_acquire() in the app. This will result in higher power consumption (by ~1mA), but will remove the glitches on GPIO36 and GPIO39.

No idea if it will affect something but is good to keep it in mind.

closed 2021-10-16 by cgreening