atomic14/diy-esp32-epub-reader #7: Better page breaking - particularly for the first section

cgreening, 2021-10-01:
There's quite a bit of pre-ample in the Gutenberg books - I think it's possible to skip this - just not sure how at the moment.

martinberlin, 2021-10-03:
I think in addition to this each page change should do a complete refresh otherwise the last page merges with the new one like you can see in this picture of the Rabbits book
[image]

- [ ] Add also page indicator below:  1 / total pages
- [ ] Save in NVS the last read page in a key/value table

**Key**: name of EPub  **Value**: last page open

so you can open the book and continue reading after resetting he device.

cgreening, 2021-10-03:
That's interesting - how did you make it do that? Was it after the device had gone to sleep and woken up? 

It's supposed to save the current state of the display so when it wakes up it knows how to update the screen - I wonder if that's not working when using spiffs.

martinberlin, 2021-10-03:
That's done simply loading the book with Spiffs and just going to next page till that one. But you can also experience it when going backwards.
If the new page are is smaller than the last page, then that part that is not drawn stays and blends with the new one.
I believe an [epd_fullclear](https://epdiy.readthedocs.io/en/latest/api.html#c.EpdiyHighlevelState) is needed between pages.

cgreening, 2021-10-03:
Very weird managed to recreate it after going into deep sleep with SPIFFS. Makes me think that the EPD hydrate and dehydrate is not working properly.

There's a clear screen at the start of the page render, but if the back buffer has not been restored properly it won't know to update the area of the screen that has old content in it.

cgreening, 2021-10-03:
For some reason the code in `EpdRenderer.h` - ` virtual void dehydrate()` isn't working... when it tries to save the back buffer it doesn't write any bytes.

```
I (36947) main: Saving state
I (36947) EPD: Dehydrating EPD
I (37477) EPD: Front buffer compressed size: 35875
I (38167) EPD: Front buffer saved 35875
I (38697) EPD: Back buffer compressed size: 35875
I (40557) EPD: Back buffer saved 0
I (40557) EPD: Dehydrated EPD
```

cgreening, 2021-10-03:
This may help - b6a0b685943ecb4fa68f0305964c2334fb55abbc

I suspect that a move to LittleFS would improve things considerably. From what I can see SPIFFS is struggling to find free space for the file so although it opens ok, it doesn't actually write any data. I've got it retrying on failure and also bumped up the SPIFFS garbage collection count.

I'll add more graceful handling when it does fail so that it gets back into a sensible state.

cgreening, 2021-10-04:
I've added this issue to move over to LittleFS - https://github.com/atomic14/diy-esp32-epub-reader/issues/13

martinberlin, 2021-10-04:
Great. When I find some time I will fork this and try to add touch.
Only show stopper is that touch consumes aprox. 1.5 mA/Hour so it kind of drains your battery. But it will be nice to have it, since with that we can have more input options, like for example a slider from 1 to last page, so you can jump to a page.

Also like I commented up there, to save last page read, will be a great usability help so you can keep on reading after deepsleep without need to navigate again to that page.

cgreening, 2021-10-04:
Also like I commented up there, to save last page read, will be a great usability help so you can keep on reading after deepsleep without need to navigate again to that page.

I'll take a look at this as it should be working.

martinberlin, 2021-10-04:
Additionally and not strictly related to this issue, there are some book cover images that are not being loaded (Maybe it's because I'm on SPIFFS?)
For example one of my favorite poetry books:
https://sync.luckycloud.de/d/bf3a5b1d14f949b89955/?p=%2FePub&mode=list
**poesia_aPizz.epub**

Other than that the book is being opening correctly and I can read it.

cgreening, 2021-10-04:
I've added an issue for the missing cover image - https://github.com/atomic14/diy-esp32-epub-reader/issues/14

It may be that the image is not compatible - progressive jpegs are not supported. But I'll investigate tonight.

cgreening, 2021-10-04:
Fixed

martinberlin, 2021-10-05:
About #14 please don't close so fast before [an additional test](https://github.com/atomic14/diy-esp32-epub-reader/issues/14#issuecomment-934081505)
It's not working for me and caught in a restart loop again.

cgreening, 2021-10-05:
Updated a comment on issue #14 - the book has some invalid utf-8 sequences which was crashing the font code in epdiy - that's now fixed upstream so should work if the git submodules update properly (having some issues with my understanding of git submodules...)

martinberlin, 2021-10-05:
Ah now. Great you are doing it good, you just have to update it locally and then push the branch like you did. 
The detail is that to update from my side I just need to do: 
$ git submodule update 
From https://github.com/atomic14/epdiy
 * branch            07925dc10e8bdf3c537270605bbe50705c9fdff2 -> FETCH_HEAD
Submodule path 'lib/epdiy/epdiy': checked out '07925dc10e8bdf3c537270605bbe50705c9fdff2'

Not: git submodule update --remote (That won't update anything)