atomic14/diy-esp32-epub-reader #11: Build instructions and first issues

martinberlin, 2021-10-02:
Hi there,
trying to compile this using PlatformIO and wanted to report first problems, so we can build proper build instructions.

1. First found, in order to hit the Build button and start with the right environment this should be added to **platformio.ini**

```
[platformio]
default_envs = lilygo_t5_47 
```
So it selects that env by default.

2. got this error when building after the default_envs setting

```
1/7] Generating esp32.ulp.ld
[2/7] Generating main.ulp.S
[3/7] Building ASM object CMakeFiles/ulp_main.dir/main.ulp.S.obj
[4/7] Linking ASM executable ulp_main
FAILED: ulp_main 
: && esp32ulp-elf-ld                                   -A elf32-esp32ulp -nostdlib CMakeFiles/ulp_main.dir/main.ulp.S.obj -o ulp_main  -T/home/martin/esp/projects/diy-esp32-epub-reader/.pio/build/lilygo_t5_47/esp-idf/src/ulp_main/esp32.ulp.ld  -Map=/home/martin/esp/projects/diy-esp32-epub-reader/.pio/build/lilygo_t5_47/esp-idf/src/ulp_main/ulp_main.map && :
esp32ulp-elf-ld: ulp_main section `.text' will not fit in region `ram'
esp32ulp-elf-ld: region `ram' overflowed by 140 bytes
ninja: build stopped: subcommand failed.
Compiling .pio/build/lilygo_t5_47/bootloader_support/src/bootloader_clock.o
Compiling .pio/build/lilygo_t5_47/bootloader_support/src/bootloader_common.o
Compiling .pio/build/lilygo_t5_47/bootloader_support/src/bootloader_flash.o
Compiling .pio/build/lilygo_t5_47/bootloader_support/src/bootloader_mem.o
Compiling .pio/build/lilygo_t5_47/bootloader_support/src/bootloader_random.o
Compiling .pio/build/lilygo_t5_47/bootloader_support/src/bootloader_random_esp32.o
Compiling .pio/build/lilygo_t5_47/bootloader_support/src/bootloader_utility.o
*** [.pio/build/lilygo_t5_47/esp-idf/src/ulp_main/ulp_main.h] Error 1
```

cgreening, 2021-10-02:
Hey Martin,

Thanks for taking a look - I really appreciate it. I thought I had replied this morning, but my reply seems to have vanished (or I forgot to save it...).

I suspect it will only work with >= 4.3 as that's what I see when I build.

```
Source `/home/martin/.platformio/packages/framework-espidf/components/soc/src/cpu_util.c' not found, needed by target `.pio/build/lilygo_t5_47/bootloader/soc/src/cpu_util.o'.
```

I think it's the same issue as this one: https://github.com/turgu1/EPub-InkPlate/issues/2#issuecomment-798987397

    SOLUTION was to delete that :
    ~/.platformio/packages$ rm -fr framework-espidf
    And build again. That recreates the folder.

I'm hoping your second issue may disappear after doing that as well - though it's not one I've seen before!

martinberlin, 2021-10-02:
Hi Chris,
did that and it did work with previous version but with this one does not. Can you try to reproduce it?

PACKAGES: 
 - framework-espidf 3.40300.0 (4.3.0) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - tool-ninja 1.7.1 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32 2.80400.210211 (8.4.0)
Reading CMake configuration...
Generating assembly for certificate bundle...
CMake Error at /home/martin/.platformio/packages/framework-espidf/tools/cmake/scripts/data_file_embed_asm.cmake:26 (file):
  file failed to open for reading (No such file or directory):

    /home/martin/esp/projects/diy-esp32-epub-reader/.pio/build/lilygo_t5_47/x509_crt_bundle

```
*** Error 1
/home/martin/.platformio/penv/bin/python /home/martin/.platformio/packages/framework-espidf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input /home/martin/.platformio/packages/framework-espidf/components/mbedtls/esp_crt_bundle/cacrt_all.pem
gen_crt_bundle.py: Parsing certificates from /home/martin/.platformio/packages/framework-espidf/components/mbedtls/esp_crt_bundle/cacrt_all.pem
gen_crt_bundle.py: Invalid certificate in /home/martin/.platformio/packages/framework-espidf/components/mbedtls/esp_crt_bundle/cacrt_all.pem
Invalid certificate
```

I know it's at this point not strictly related to your code but is a real show stopper, since you cannot build it!

cgreening, 2021-10-02:
Tried deleting everything in `~/.platformio/packages` and it redownloaded everything when I hit build.

Here's what I see when I hit build:

```
PACKAGES: 
 - framework-espidf 3.40300.0 (4.3.0) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - tool-ninja 1.9.0 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32 2.80400.210211 (8.4.0)
WARNING: You are using pip version 21.0.1; however, version 21.2.4 is available.
You should consider upgrading via the '/Users/chrisgreening/.platformio/penv/bin/python -m pip install --upgrade pip' command.
Reading CMake configuration...
Generating assembly for certificate bundle...
```

And then it carries on without any problems :(

cgreening, 2021-10-02:
I'm running on a mac and have python 3.9.7 - I wonder if that's the difference?

cgreening, 2021-10-02:
I've tried switching off the certificate bundle in menuconfig - can you pull the latest code and see if that fixes it?

martinberlin, 2021-10-02:
Thanks Chris,
like this it builds. But it shouldn't be like that that you need to disable the certificate bundle. I don't really know what consequences has it, but good thing is that it compiles ;)

Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   9.8% (used 32180 bytes from 327680 bytes)
Flash: [========  ]  81.8% (used 857325 bytes from 1048576 bytes)
Building .pio/build/lilygo_t5_47/firmware.bin
esptool.py v3.1
Merged 1 ELF section
=================================================== [SUCCESS] Took 38.41 seconds

cgreening, 2021-10-02:
That's great! Should have minimal impact unless we need to connect to outside world over https. Hopefully by then the problem will have been fixed :)

Good point on the ports - I'll get those removed.

martinberlin, 2021-10-02:
Awesome. It works now. Please add also on the Readme that for the:

    USE_SPIFFS

version you need to run:

    pio run -t uploadfs

in order to upload the /data folder to your Spiffs. Notice that also when building they come some warnings (Not your library but still...)

```
lib/png/PNGdec/src/infback.c: In function 'inflateBack':
lib/png/PNGdec/src/infback.c:479:25: warning: this statement may fall through [-Wimplicit-fallthrough=]
             state->mode = LEN;
             ~~~~~~~~~~~~^~~~~
lib/png/PNGdec/src/infback.c:481:9: note: here
         case LEN:
         ^~~~
Indexing .pio/build/lilygo_t5_47/bootloader/esp-idf/spi_flash/libspi_flash.a
Compiling .pio/build/lilygo_t5_47/libc97/png/inftrees.o
Linking .pio/build/lilygo_t5_47/bootloader.elf
lib/png/PNGdec/src/inflate.c: In function 'inflate':
lib/png/PNGdec/src/inflate.c:750:25: warning: this statement may fall through [-Wimplicit-fallthrough=]
             state->mode = EXLEN;
             ~~~~~~~~~~~~^~~~~~~
lib/png/PNGdec/src/inflate.c:751:9: note: here
         case EXLEN:
         ^~~~
lib/png/PNGdec/src/inflate.c:763:25: warning: this statement may fall through [-Wimplicit-fallthrough=]
             state->mode = EXTRA;
             ~~~~~~~~~~~~^~~~~~~
lib/png/PNGdec/src/inflate.c:764:9: note: here
         case EXTRA:
         ^~~~
lib/png/PNGdec/src/inflate.c:785:25: warning: this statement may fall through [-Wimplicit-fallthrough=]
             state->mode = NAME;
             ~~~~~~~~~~~~^~~~~~
lib/png/PNGdec/src/inflate.c:786:9: note: here
         case NAME:
         ^~~~
lib/png/PNGdec/src/inflate.c:806:25: warning: this statement may fall through [-Wimplicit-fallthrough=]
             state->mode = COMMENT;
```

Thanks a lot looks very cool. Soon I will upload a video in Twitter!

cgreening, 2021-10-02:
Fantastic!

I'll update the readme with what you've found tomorrow.

Thanks for being the first to try it and go through the pain!

martinberlin, 2021-10-03:
No problem, it has potential to be a very nice ebook reader!
Thanks for your help on this one

closed 2021-10-03 by martinberlin

alvaro-martin, 2021-11-07:
For this kind of error:
.pio/build/lilygo_t5_47/x509_crt_bundle
What it works for me (I'm working on linux-based system) is going to this file:
[image]
Search specifically for this certificate "EC-ACC" and delete it:
[image]

cgreening, 2021-11-07:
That's great - at the moment we've just disabled certificate bundling - but that's not really a long term solution...

martinberlin, 2021-11-07:
Yes this is a good option. I did it do I don't have to worry anymore. But at some point it will kick back on us if we add some download option that requires secure https