atomic14/diy-esp32-epub-reader #12: Restart loop when a file is not found in ePub

martinberlin, 2021-10-03:
Hi Chris,
I found this using this book that I you can download here "El Aleph":
https://freeditorial.com/es/books/el-aleph

In Serial I see the following output:

```
I (804) Controls: ULP Wakeup
I (2164) main: Creating epub list
E (3234) ZIP: Could not find file OEBPS/content.opf
E (3234) EPUB: Error parsing content.opf - XML_ERROR_EMPTY_DOCUMENT
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
```

In my opinion it should not restart when it does not find a related file. Note that I got this restart loop at start when it's making the list, not when opening the book.

cgreening, 2021-10-04:
I've assumed that all the files in the epub archive are in a folder called "OEBPS/" - but it looks like that isn't always the case. The example you've found just has the files in the root of the archive.

Should be straightforward enough to allow for that.

cgreening, 2021-10-04:
Looks like I should be looking in `META_INFO/container.xml`

I'll fix that tonight.

```
<?xml version="1.0"?>
<container version="1.0" xmlns="urn:oasis:names:tc:opendocument:xmlns:container">
   <rootfiles>
      <rootfile full-path="content.opf" media-type="application/oebps-package+xml"/> 
   </rootfiles>
</container>
```

closed 2021-10-04 by cgreening