atomic14/diy-esp32-epub-reader #13: Switch over to LittleFS

cgreening, 2021-10-04:
SPIFFS seems to be a bit broken (doesn't always seem to be able to save data).

Let's try switching to LittleFS: https://github.com/joltwallet/esp_littlefs

cgreening, 2021-10-05:
This seems to be a lot more complicated than I initially thought. There isn't official support for LittleFS in the ESP32 and platformio doesn't have out of the box support for building LittleFS filesystems when using the ESP32.

Pablo2048, 2021-10-05:
Hi Chris. There is official support for LittleFS, unfortunately in Core 2.0.0 which is not availble right now directly in pio. You can do some workarounds. For example:
  * You can use Lorol's LITTLEFS library (Library name has changed in the Core for ESP8266 compatibility)
  * You can use small patch to filesystem build tools. The patch consists of two steps: one is simple python script:
 ```python
Import("env")
env.Replace( MKSPIFFSTOOL=env.get("PROJECT_DIR") + '/tools/mklittlefs/mklittlefs' )
```
the script is named `littlefsbuilder.py`, and I'm using it like this (in the platformio.ini):
```
extra_scripts =
  ./tools/littlefsbuilder/littlefsbuilder.py
```
You have to install `mklittlefs` tool into tools/mklittlefs/ subdirectory.

You can give it a try.

cgreening, 2021-10-18:
Tempted to wait on this one until platformio has official support - I'm a bit worried about how to make it work cross platform.