atomic14/diy-esp32-epub-reader #14: Missing cover image from book

cgreening, 2021-10-04:
See this epub file:

https://sync.luckycloud.de/d/bf3a5b1d14f949b89955/?p=%2FePub&mode=list

Data all looks correct - but no cover image.

martinberlin, 2021-10-04:
Nice one I will test this tomorrow!

martinberlin, 2021-10-05:
This should not be closed yet. After yestersday's update is caught in a restart loop:

```
I (515) main: Battery 100, 3894.00v
I (515) main: Memory before renderer init: 4401791
I (585) main: Memory after renderer init: 3258507
I (585) main: Using SPIFFS
I (585) SPIFFS: Initializing SPIFFS
I (805) main: Memory after sdcard init: 3255583
I (805) Controls: ULP Wakeup
I (2165) main: Creating epub list
I (4025) main: Epub files loaded
I (4025) PUBLIST: Current page is 0, previous page -1, redraw=0
I (4225) PUBLIST: Rendering item 0
I (4425) EPUB: Reading item OEBPS/Images/cover.jpg
I (4895) JPG: JPEG Decoded - size 600,900, scale = 0.640000, 0.644444, 2
E (5505) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (5505) task_wdt:  - IDLE (CPU 1)
E (5505) task_wdt: Tasks currently running:
E (5505) task_wdt: CPU 0: IDLE
E (5505) task_wdt: CPU 1: main_task
E (5505) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x40116F4F:0x3FFB0840 0x40084209:0x3FFB0860 0x40137E57:0x3FFBBAF0 0x40116CB3:0x3FFBBB10 0x4008F2C1:0x3FFBBB30 0x40090A21:0x3FFBBB50v
```

martinberlin, 2021-10-05:
Sure!

     $ git submodule update --remote
     Submodule path 'lib/epdiy/epdiy': checked out '8b4302a6a650a5c228520c100b862aba60978c2c'

A more detailed Serial output of the error:

```
I (6685) EPUB: Reading item OEBPS/Images/cover.jpg
I (7065) JPG: JPEG Decoded - size 600,900, scale = 0.640000, 0.644444, 2
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400dd9c9  PS      : 0x00060e30  A0      : 0x800d2816  A1      : 0x3ffcc530  
A2      : 0x3f41b220  A3      : 0x3ffcc60e  A4      : 0x000000fd  A5      : 0x00000006  
A6      : 0x0000003f  A7      : 0x00000006  A8      : 0x000000fd  A9      : 0x00000005  
A10     : 0x0000f007  A11     : 0x3ffb0094  A12     : 0x00000021  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0x00000063  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000f007  LBEG    : 0x400dd8d6  LEND    : 0x400dd8f6  LCOUNT  : 0x00000002  

Backtrace:0x400dd9c6:0x3ffcc530 0x400d2813:0x3ffcc580 0x400e037d:0x3ffcc5d0 0x400dfa21:0x3ffcc680 0x400d29b6:0x3ffcc700 0x400d2a6c:0x3ffcc720 0x400d2d22:0x3ffcc740 0x40090a21:0x3ffcc790
  #0  0x400dd9c6:0x3ffcc530 in utf8_len at lib/epdiy/epdiy/src/epd_driver/font.c:46
      (inlined by) next_cp at lib/epdiy/epdiy/src/epd_driver/font.c:62
      (inlined by) epd_get_text_bounds at lib/epdiy/epdiy/src/epd_driver/font.c:248
  #1  0x400d2813:0x3ffcc580 in EpdRenderer::get_text_width(char const*, bool, bool) at lib/Epub/Renderer/EpdRenderer.h:92
  #2  0x400e037d:0x3ffcc5d0 in Renderer::draw_text_box(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, int, int, int, bool, bool) at lib/Epub/Renderer/Renderer.cpp:67
  #3  0x400dfa21:0x3ffcc680 in EpubList::render() at lib/Epub/EpubList/EpubList.cpp:184
  #4  0x400d29b6:0x3ffcc700 in handleEpubList(Renderer*, UIAction) at src/main.cpp:119
  #5  0x400d2a6c:0x3ffcc720 in handleUserInteraction(Renderer*, UIAction) at src/main.cpp:131
  #6  0x400d2d22:0x3ffcc740 in main_task(void*) at src/main.cpp:212
  #7  0x40090a21:0x3ffcc790 in vPortTaskWrapper at /home/martin/.platformio/packages/framework-espidf/components/freertos/port/xtensa/port.c:168
```

If I see this correctly is hanging at this line:

```c
  while ((c = next_cp((const uint8_t **)&string))) {   // <-- HERE
    get_char_bounds(font, c, &temp_x, &temp_y, &minx, &miny, &maxx, &maxy, &props);
  }
```

But that's strange before it didn't hang with this book on font.c, just the cover was missing.

martinberlin, 2021-10-05:
Awesome, now it works and the cover of this book is displayed properly, thanks!!!

cgreening, 2021-10-05:
I really need to try and understand git submodules better.

closed 2021-10-05 by cgreening