atomic14/diy-esp32-epub-reader #18: Task Watchdog triggered

cgreening, 2021-10-05:
Seeing this a lot particularly when rendering large jpegs.

martinberlin, 2021-10-05:
Hi Chris,
I also experienced this but however I was able to render up to 600 Kb JPGs without resets. A WDT alert should not mean that ESP32 resets. Can you try to feed the watchdog?'
Check [this commit](https://github.com/martinberlin/cale-idf/commit/d07246808542dc784fa8055699d219255b11d503#diff-41ae480505ef2ecc3ddb4f56895d1e4c5dc17d0b1f72ced73d2b657b57d386d6R281) when I was experiencing the same with my JPG render.

martinberlin, 2021-10-06:
This is my next show-stopper. Since I had this issue with the broken button 35, every time I tap to select a book, it hangs on this WDT alert (And I don't know why)