atomic14/diy-esp32-epub-reader #20: Adding experimental touch

martinberlin, 2021-10-05:
Hi Chris,
started a fork here: https://github.com/martinberlin/diy-esp32-epub-reader-touch

with the mission to add touch (If I can, really experimental)
And I wanted to leave this topic to ask you about building an ESP-IDF project with Platformio since it's my first time and for sure I will make lot's of mistakes. 
For example in [this commit](https://github.com/martinberlin/diy-esp32-epub-reader-touch/commit/d1474edf4a8e26fb38b44704eee71398f86311c5) just added my touch component in **lib** folder. 
But I'm still trying to guess how I tell Platformio to include this in the build process. In IDF I do that using CMakeFiles in the component folder but here I'm a bit lost. Can you share your knowledge with me?
Thanks in advance

cgreening, 2021-10-05:
With the lib folder, platformio automatically includes it as a dependency so you don't need to do anything.

With the current version of platformio, you can also just create a `components` folder and use it just as you would with the esp-idf - it should work out what to do with it.

If you want I can make you a contributor to this repo.

cgreening, 2021-10-05:
I think since your touch libary is already an idf component then going the `components` folder route might be easier. This will also wire up any menuconfig settings you have as well. You can launch menu config with

```
platformio run -t menuconfig
```

martinberlin, 2021-10-05:
I tried with the touch library adding it on the lib folder but it does not include it automatically as a dependance for me. I wanted to follow your path and leave all the libs on the same folder but if I cannot get it to work I will do as you suggest. Thanks anyways!
I would like to discover why other libraries are added automatically but mine not ;)

Let's see if I can really collaborate with something first, then I can accept your invitation!

martinberlin, 2021-10-06:
Found what it is. If the header includes are in a directory, then you need to add the following library.json

```json
{
  "build": {
    "includeDir": "include",
    "srcDir": "."
  }
}
```

And only then it will be included!