atomic14/diy-esp32-epub-reader #21: Optimization: Making the display flush faster and other ideas

martinberlin, 2021-10-05:
Hi Chris, this is working already great but I have some ideas that I wanted to share with you. 
First I'm suspecting that most of the pages are rendering with MODE_GC16 update mode that is not the faster if you need to render only black/white text.

Adding a Serial output here: 

```c
void flush_display()
  {
    printf("needs_gray:%d\n", needs_gray_flush);
    epd_hl_update_screen(&m_hl, needs_gray_flush ? MODE_GC16 : MODE_DU, 20);
    needs_gray_flush = false;
  }
```

You can see that even on pages where is pure text, needs_gray is always true. This should be corrected since it will make navigating pages much faster!

Another optimization tip can be to use a smaller hourglass (Maybe half the size) that will also help to be a bit faster. But mainly the MODE_GC16 should be only used on the pages that render images and need grayscale.

As a resume:

- [ ] MODE_GC16 only on pages that require it
- [ ] Smaller visual indicator when a button is pressed (optional)
- [ ] Battery indicator is shown always full for me even if it's almost no power left (last wish, not important)

cgreening, 2021-10-05:
I took a look at that before and found that the text seems to be anti-alised and switching off the GC16 mode made it look a bit weird. If you have a look at the font files you can see that the bitmaps aren't always 00 of FF but have values in between.

I do wonder how much difference it actually makes - the display is quite high resolution so maybe the antialiasing is not adding that much? We could probably hack the epdiy code to draw the fonts in black and white.

I've noticed that issue with the battery indicator as well - I've split it out to a new issue: https://github.com/atomic14/diy-esp32-epub-reader/issues/22

Agreed on the indicator - it's bit horrible - I wonder if just showing something up in the header where the battery indicator is would be more subtle.

martinberlin, 2021-10-05:
Agree on your view. Fonts do not look so nice, but as a benefit, navigating the pages updating this line to use only MODE_DU is much nicer.
I think there should be a way to make the fonts only black / white, maybe let's ask in the epdiy Slack?   For sure other libraries I used, like Adafruit GFX, render the fonts only in black and white without any problem since they are used in monochorome displays.
But not a problem, let's leave it like it is now, and I up-vote to move to a smaller hourglass placed maybe at the center of the top bar (Or at the right near the battery indicator?)

cgreening, 2021-10-05:
Might be an option that can added to the draw string functions of epdiy.

cgreening, 2021-10-05:
Fixed with b2385362a4a6e426bc761db2225b4d2554e4d30f

Screen updates are much faster when changing pages - when there is only text on the screen.

closed 2021-10-05 by cgreening