atomic14/diy-esp32-epub-reader #24: Update to latest epdiy restarts

martinberlin, 2021-10-06:
If you do a: git submodules update --remote then epiy get's last update:

epdiy @ 8b4302a 

Currently is: 
epdiy @ 07925dc   -> Works

Updating to the latest generates this backtrace:

```
Backtrace:0x400dda2a:0x3ffcc550 0x400d281f:0x3ffcc5a0 0x400e03d9:0x3ffcc5f0 0x400dfa85:0x3ffcc6a0 0x400d29e6:0x3ffcc720 0x400d2a9c:0x3ffcc740 0x400d2d66:0x3ffcc760 0x40090a21:0x3ffcc7c0
  #0  0x400dda2a:0x3ffcc550 in utf8_len at lib/epdiy/epdiy/src/epd_driver/font.c:46
      (inlined by) next_cp at lib/epdiy/epdiy/src/epd_driver/font.c:62
      (inlined by) epd_get_text_bounds at lib/epdiy/epdiy/src/epd_driver/font.c:248
```

cgreening, 2021-10-06:
That's weird - I think the latest epdiy is 07925dc10e8bdf3c537270605bbe50705c9fdff2

I wonder why you ending up with 8b4302a

cgreening, 2021-10-06:
In your branch it might be pointing to old commit - `git submodules update` is a bit confusing - it doesn't get the latest code from the submodule, it just makes sure the submodule is pointing at the commit that it thinks it should be at.

To change the commit the submodule is pointing at you need to do the following:

```
cd lib/epdiy/epdiy
git checkout 07925dc10e8bdf3c537270605bbe50705c9fdff2
```

If that's worked then you should see a change in the main repository that you can commit.

martinberlin, 2021-10-06:
By the way the book that generates this is the one I referenced before:
https://github.com/atomic14/diy-esp32-epub-reader/blob/main/data/poesia_aPizz.epub

With the others work correctly. That is the commit where the submodules update --remote goes. No idea why, but anyways the last one is just a Readme update so I think is ok anyways since I'm using latest version.

cgreening, 2021-10-06:
That's the book with the broken utf8 sequence which should be fixed in the latest epdiy commit.

martinberlin, 2021-10-06:
Great solved. You are right submodules --update is kind of weird. 
But I'm sure I worked like that in other projects without any problem. Thanks now it works.
If you want to test Touch, you can already see coordinates and events:

```
X:505 Y:192 E:3
X:505 Y:192 E:0
e 4 1  X: 348 Y: 455 E: tap
X:505 Y:192 E:0
```

In my latest stand in main branch (Was not master before the default main branch?)
https://github.com/martinberlin/diy-esp32-epub-reader-touch

closed 2021-10-06 by martinberlin