atomic14/diy-esp32-epub-reader #29: Touch thoughts

cgreening, 2021-10-07:
I've done a bit of refactoring on the touch code - not sure if it's really correct yet.

A couple of thoughts I had.

It's quite hard to hit the buttons - we could:

When reading a book, go to the next page by tapping the right of the screen and go left by tapping the left of the screen. Get back to the main menu by tapping near the top of the screen.

When selecting books got up and down by tapping the top and bottom of the screen and select by tapping in the centre - not sure how intuitive that is though...

Is it possible with the touch library to detect a "touch up" ie the end of the touch interaction when the finger is removed from the screen? That would be a good point to kick off the interaction.

I need to pull a lot of the code out of the main function so that it's a bit easier to drive the UI from controls. I'll make a start on that next.

cgreening, 2021-10-07:
Actually, for browsing the books a fairly large arrow at the top and bottom of the screen would let you page through the books nicely.

martinberlin, 2021-10-08:
Nice job. There is an additional thing to do on my component side:

```c
// Updating this in the L58Touch::processTouch to 50 millis instead of 150:
if (tapDetectionEnabled && _touchEndTime - _touchStartTime <= 50) {
        fireEvent(point, TEvent::Tap);
    }
```

Makes it much smoother and reduces the double click effect. I will convert that into a variable that can be set on class instantiation, setTapMillis or setTapThreshold so we can modify it. 
Notice I'm getting this sometimes since yesterday update (No idea if it's related or a casual problem with SPIFFS)

I (526) main: Controls configured
I (526) Controls: ULP Wakeup
I (536) EPD: Hydrating EPD
I (686) EPD: Buffer compressed size: 40567
E (866) EPD: No data to restor
I (866) EPD: Hydrate EPD failed

Which results in the list being rendered blank.

cgreening, 2021-10-08:
I think this is good enough now.

Your RTC error is due to using using GPIO_NUM_1 instead of GPIO_NUM_35 - make sure you've got the correct commit of epdiy in the submodule as the latest version fixes the always low problem.

closed 2021-10-08 by cgreening