atomic14/diy-esp32-epub-reader #66: low battery on M5Paper causes reboot cycle

cgreening, 2021-11-08:
I'm guessing that below a certain level the SDCard stops working causing a crash.

Maybe related to https://github.com/atomic14/diy-esp32-epub-reader/issues/62

martinberlin, 2021-11-10:
I think also IT8951 needs more power as a controller than the EPDiy based devices. That could be also a factor for this crash.
Try to measure how much it consumes LilyGo vs M5 to see if there is a difference.

lypanov, 2023-05-12:
This happens on my M5Paper even at a very high battery level FWIW.

DonCarnohan, 2023-08-02:
I was also having this issue with a fully charged battery
I think I was able to narrow this issue down to the sd card. I had a spare UHS (30MB/s) card in and it was causing a brownout:

```
I (135006) main: Entering deep sleep
I (452) psram: This chip is ESP32-D0WD
I (453) spiram: Found 64MBit SPI RAM device
I (453) spiram: SPI RAM mode: flash 80m sram 80m
I (455) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (463) cpu_start: Pro cpu up.
I (466) cpu_start: Starting app cpu, entry point is 0x40082ca0
... etc... etc...
I (616) main: Powering up the board
I (616) main: Creating renderer
I (636) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr�[0m
I (636) gpio: GPIO[27]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0

Brownout detI (464) psram: This chip is ESP32-D0WD
I (464) spiram: Found 64MBit SPI RAM device
I (465) spiram: SPI RAM mode: flash 80m sram 80m
I (467) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (474) cpu_start: Pro cpu up.
I (478) cpu_start: Starting app cpu, entry point is 0x40082ca0
```

This would only happen when waking from deep sleep while in `READING_EPUB` state. While selecting EPUB or TOC it would wake from deep sleep fine...

I've since put a lower speed 10MB/s card and no longer have this issue. Not exactly an answer but if anyone has some tips to help me dig in further, it'd be much appreciated. Even if the tip is that it's probably not worth digging and is likely a hardware limitation...