atomic14/diy-esp32-epub-reader #69: Improve and refactor `EpubIndex`

cgreening, 2021-11-17:
Rename to `EpubToc` and make the cell height adjustable so that it's more compact

martinberlin, 2021-11-18:
Related to this, when rendering the Toc, we should add a left margin of 10px or more to the list otherwise the selection box touches the font.

martinberlin, 2021-11-20:
@cgreening additional weird bug here. Opening first index, usually cover, and then pressing previous page button generates a never ending loop of madness.
As a normal behavior before implementing this it went back to List view. 
Question is what it should do now? Go back to TOC view?

cgreening, 2021-11-20:
Yeah, should probably go back to the TOC view.

We could have the `prev` and `next` functions return true or false depending on if they succeed - and then trigger navigation based on that.

martinberlin, 2021-11-20:
@cgreening check the last commit, probably you will do it more elegantly, but now it does not reboot forever just because there is a vector that goes out of range.

martinberlin, 2021-11-23:
>make the cell height adjustable

@cgreening about this nice thought, what will be your idea to implement this?
Because if I'm thinking this right, then we will need to store what is the height of each cell for each TOC item, or taht is not needed?
It will also make more complicated the UX because the rectangle select box needs to adapt to this right?   It sounds complicated at least in the first round of thought.
But maybe you see it differently.

closed 2021-11-30 by cgreening