atomic14/diy-esp32-epub-reader #73: Opening certain part of TOC crashes

martinberlin, 2021-11-21:
To reproduce open Chapter VIII of Blade runners "Film noir" book.

```
I (27829) EREADER: go to section:15
I (28129) EPUB: toc path: OEBPS/9780062852892_toc.ncx

I (28289) EREADER: Parse and render section 15
I (28289) EREADER: item OEBPS/text/9780062852892_Chapter_8.xhtml

abort() was called at PC 0x40112937 on core 1
TRACE relevant parts show: 
Looks if there is some problem with the itemSize parsing the XML. Previous chapters before 8 open just fine...

  #15 0x4014568f:0x3ffd0010 in tinyxml2::MemPoolT<52>::ItemSize() const at .pio/libdeps/epdiy/tinyxml2/tinyxml2.h:370
  #16 0x40145607:0x3ffd0030 in tinyxml2::XMLText::Accept(tinyxml2::XMLVisitor*) const at .pio/libdeps/epdiy/tinyxml2/tinyxml2.cpp:1251
  #17 0x400e538d:0x3ffd0050 in RubbishHtmlParser::parse(char const*, int) at lib/Epub/RubbishHtmlParser/RubbishHtmlParser.cpp:179

      (inlined by) RubbishHtmlParser::RubbishHtmlParser(char const*, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at lib/Epub/RubbishHtmlParser/RubbishHtmlParser.cpp:58
  #19 0x400e3a41:0x3ffd0260 in EpubReader::parse_and_layout_current_section() at lib/Epub/EpubList/EpubReader.cpp:55 (discriminator 3)
  #20 0x400e3b27:0x3ffd02c0 in EpubReader::set_state_section(unsigned short) at lib/Epub/EpubList/EpubReader.cpp:104
  #21 0x400d3b5a:0x3ffd02e0 in handleEpub(Renderer*, UIAction) at src/main.cpp:83

```

cgreening, 2021-11-30:
Hmm, could be running out of memory - I wonder if it's a particularly heavy chapter?

martinberlin, 2021-12-08:
[image]
Good guess. Yes 9780062852892_Chapter_8.xhtml is with a difference the largest Chapter in this book.
So it will need 290 Kb or memory to parse it. This can be a problem in some books that have too much content in a single xhtml.

martinberlin, 2021-12-09:
@cgreening tell me if you can reproduce this one. A funny fact is that this happens if you open Chapter 8 directly from the TOC. But if you navigate from Chapter 7 to 8 this does not happen. Would like to try catching the error instead of generating a restart loop. 

What would be the best way to get out of this restart loops? They are really annoying and there is no other way than a hardware reset. 

Readme update preview: https://github.com/atomic14/diy-esp32-epub-reader/tree/feature/better-error-handling
Added also links in the footer both in Readme and Wiki.

commit e82dab46 (2021-12-09): Fixes #73
closed 2022-03-01 by cgreening