atomic14/diy-esp32-epub-reader #74: m5paper compile error

sn0wst0rm, 2021-11-26:
Hi, I was trying to compile and test your project on my m5paper, but I can't get it to work.
After cloning it, I selected the correct env task for m5paper, but it gave me a lot of compilation errors:

`Compiling .pio/build/m5_paper/app_trace/app_trace.o
In file included from src/boards/Lilygo_t5_47.cpp:2:
lib/Fonts/regular_font.h:2:10: fatal error: epd_driver.h: No such file or directory

********************************************************************
* Looking for epd_driver.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:epd_driver.h"
* Web  > https://platformio.org/lib/search?query=header:epd_driver.h
*
********************************************************************

 #include "epd_driver.h"
          ^~~~~~~~~~~~~~
compilation terminated.
Compiling .pio/build/m5_paper/src/boards/controls/EpdiyV6ButtonControls.o
*** [.pio/build/m5_paper/src/boards/Lilygo_t5_47.o] Error 1
Compiling .pio/build/m5_paper/src/boards/controls/GPIOButtonControls.o
In file included from src/boards/controls/EpdiyV6ButtonControls.cpp:1:
src/boards/controls/EpdiyV6ButtonControls.h:2:10: fatal error: pca9555.h: No such file or directory

*****************************************************************
* Looking for pca9555.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pca9555.h"
* Web  > https://platformio.org/lib/search?query=header:pca9555.h
*
*****************************************************************

 #include "pca9555.h"
          ^~~~~~~~~~~
compilation terminated.
*** [.pio/build/m5_paper/src/boards/controls/EpdiyV6ButtonControls.o] Error 1
In file included from lib/Epub/Renderer/M5PaperRenderer.h:6,
                 from src/boards/M5Paper.cpp:2:
lib/Epub/Renderer/EpdiyFrameBufferRenderer.h:3:10: fatal error: epd_driver.h: No such file or directory

********************************************************************
* Looking for epd_driver.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:epd_driver.h"
* Web  > https://platformio.org/lib/search?query=header:epd_driver.h
*
********************************************************************

 #include <epd_driver.h>
          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/m5_paper/src/boards/M5Paper.o] Error 1
In file included from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:53:6: error: conflicting declaration of 'DIR* opendir(const char*)' with 'C' linkage
 DIR *opendir(const char *);
      ^~~~~~~
In file included from /Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:39,
                 from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/framework-espidf/components/newlib/platform_include/sys/dirent.h:53:6: note: previous declaration with 'C++' linkage
 DIR* opendir(const char* name);
      ^~~~~~~
In file included from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:56:3: error: conflicting declaration of 'dirent* readdir(DIR*)' with 'C' linkage
   readdir(DIR *);
   ^~~~~~~
In file included from /Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:39,
                 from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/framework-espidf/components/newlib/platform_include/sys/dirent.h:54:16: note: previous declaration with 'C++' linkage
 struct dirent* readdir(DIR* pdir);
                ^~~~~~~
In file included from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:58:6: error: conflicting declaration of 'int readdir_r(DIR*, dirent*, dirent**)' with 'C' linkage
 int  readdir_r(DIR *__restrict, struct dirent *__restrict,
      ^~~~~~~~~
In file included from /Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:39,
                 from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/framework-espidf/components/newlib/platform_include/sys/dirent.h:59:5: note: previous declaration with 'C++' linkage
 int readdir_r(DIR* pdir, struct dirent* entry, struct dirent** out_dirent);
     ^~~~~~~~~
In file included from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:61:7: error: conflicting declaration of 'void rewinddir(DIR*)' with 'C' linkage
 void  rewinddir(DIR *);
       ^~~~~~~~~
In file included from /Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:39,
                 from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/framework-espidf/components/newlib/platform_include/sys/dirent.h:57:6: note: previous declaration with 'C++' linkage
 void rewinddir(DIR* pdir);
      ^~~~~~~~~
In file included from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:72:7: error: conflicting declaration of 'void seekdir(DIR*, long int)' with 'C' linkage
 void  seekdir(DIR *, long);
       ^~~~~~~
In file included from /Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:39,
                 from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/framework-espidf/components/newlib/platform_include/sys/dirent.h:56:6: note: previous declaration with 'C++' linkage
 void seekdir(DIR* pdir, long loc);
      ^~~~~~~
In file included from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:73:7: error: conflicting declaration of 'long int telldir(DIR*)' with 'C' linkage
 long  telldir(DIR *);
       ^~~~~~~
In file included from /Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:39,
                 from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/framework-espidf/components/newlib/platform_include/sys/dirent.h:55:6: note: previous declaration with 'C++' linkage
 long telldir(DIR* pdir);
      ^~~~~~~
In file included from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:76:6: error: conflicting declaration of 'int closedir(DIR*)' with 'C' linkage
 int  closedir(DIR *);
      ^~~~~~~~
In file included from /Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:39,
                 from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/framework-espidf/components/newlib/platform_include/sys/dirent.h:58:5: note: previous declaration with 'C++' linkage
 int closedir(DIR* pdir);
     ^~~~~~~~
*** [.pio/build/m5_paper/src/main.o] Error 1
================================================================================== [FAILED] Took 6.32 seconds ==================================================================================

Environment    Status    Duration
-------------  --------  ------------
m5_paper       FAILED    00:00:06.320`

I noticed that in platform.ini config file, in the m5paper env section, there was not specified anywhere the m5stack/M5EPD library. I tried adding it as a dependency but as I thought it did not work because it is written to be built against the Arduino framework, while you use espidf.

Can you please tell me how you did get it to compile? It would be really appreciated!

cgreening, 2021-11-26:
Make sure you use `git clone --recursive ...`

It sounds like you haven't got all the sub modules so it can't find a lot of things.

sn0wst0rm, 2021-11-26:
Hi Chris, thanks for all the work you are doing and for the help provided. However I did clone recursively the repository, but to be sure I deleted and recloned it one more time.

This time the are other compile errors:
```
Compiling .pio/build/m5_paper/app_trace/gcov/gcov_rtio.o
In file included from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:53:6: error: conflicting declaration of 'DIR* opendir(const char*)' with 'C' linkage
 DIR *opendir(const char *);
      ^~~~~~~
In file included from /Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:39,
                 from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/framework-espidf/components/newlib/platform_include/sys/dirent.h:53:6: note: previous declaration with 'C++' linkage
 DIR* opendir(const char* name);
      ^~~~~~~
In file included from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:56:3: error: conflicting declaration of 'dirent* readdir(DIR*)' with 'C' linkage
   readdir(DIR *);
   ^~~~~~~
In file included from /Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:39,
                 from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/framework-espidf/components/newlib/platform_include/sys/dirent.h:54:16: note: previous declaration with 'C++' linkage
 struct dirent* readdir(DIR* pdir);
                ^~~~~~~
In file included from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:58:6: error: conflicting declaration of 'int readdir_r(DIR*, dirent*, dirent**)' with 'C' linkage
 int  readdir_r(DIR *__restrict, struct dirent *__restrict,
      ^~~~~~~~~
In file included from /Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:39,
                 from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/framework-espidf/components/newlib/platform_include/sys/dirent.h:59:5: note: previous declaration with 'C++' linkage
 int readdir_r(DIR* pdir, struct dirent* entry, struct dirent** out_dirent);
     ^~~~~~~~~
In file included from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:61:7: error: conflicting declaration of 'void rewinddir(DIR*)' with 'C' linkage
 void  rewinddir(DIR *);
       ^~~~~~~~~
In file included from /Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:39,
                 from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/framework-espidf/components/newlib/platform_include/sys/dirent.h:57:6: note: previous declaration with 'C++' linkage
 void rewinddir(DIR* pdir);
      ^~~~~~~~~
In file included from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:72:7: error: conflicting declaration of 'void seekdir(DIR*, long int)' with 'C' linkage
 void  seekdir(DIR *, long);
       ^~~~~~~
In file included from /Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:39,
                 from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/framework-espidf/components/newlib/platform_include/sys/dirent.h:56:6: note: previous declaration with 'C++' linkage
 void seekdir(DIR* pdir, long loc);
      ^~~~~~~
In file included from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:73:7: error: conflicting declaration of 'long int telldir(DIR*)' with 'C' linkage
 long  telldir(DIR *);
       ^~~~~~~
In file included from /Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:39,
                 from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/framework-espidf/components/newlib/platform_include/sys/dirent.h:55:6: note: previous declaration with 'C++' linkage
 long telldir(DIR* pdir);
[... 1 lines cut ...]
In file included from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:76:6: error: conflicting declaration of 'int closedir(DIR*)' with 'C' linkage
 int  closedir(DIR *);
      ^~~~~~~~
In file included from /Users/sn0wst0rm/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:39,
                 from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/sn0wst0rm/.platformio/packages/framework-espidf/components/newlib/platform_include/sys/dirent.h:58:5: note: previous declaration with 'C++' linkage
[... cut ...]
========================================================================================== [FAILED] Took 11.40 seconds ==========================================================================================

Environment    Status    Duration
-------------  --------  ------------
m5_paper       FAILED    00:00:11.402
===================================================================================== 1 failed, 0 succeeded in 00:00:11.402 =====================================================================================
The terminal process "platformio 'run', '--environment', 'm5_paper'" terminated with exit code: 1.
```

sn0wst0rm, 2021-11-26:
@cgreening regarding the m5paper EPD driver, how did you get it working without using the framework provided by m5stack? I don't really like developing with the Arduino Framework so if some port for the espressif framework is available I'd like to test it.

Thanks for the help!

cgreening, 2021-11-26:
I ported over the minimum required to send a full frame buffer over. Seems
to work ok.

I’m away from the computer for the weekend, but the code should build as we
run a GitHub action to check in every commit.

I’ll investigate Monday when I’m back in.

cgreening, 2021-11-28:
Just tried to recreate the issue - the build is working for me locally. I wonder if it's some issue with the platformio/esp-idf version. Here's what I see when I start a build off:

```
> Executing task: platformio run --environment m5_paper <

Processing m5_paper (platform: espressif32; board: esp32dev; framework: espidf)
-------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.3.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf 3.40300.0 (4.3.0) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - tool-ninja 1.9.0 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32 2.80400.210211 (8.4.0)
```

titimoby, 2021-11-29:
In order to contribute on the analysis on compile errors, here is my start block configuration on macOS and then the errors
```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.4.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf 3.40301.0 (4.3.1) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - tool-ninja 1.9.0 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-riscv32-esp 8.4.0+2021r1 
 - toolchain-xtensa-esp32 8.4.0+2021r1 
 - toolchain-xtensa-esp32s2 8.4.0+2021r1
```

A succession of this kind of errors:

```
In file included from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/thierry/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:53:6: error: conflicting declaration of 'DIR* opendir(const char*)' with 'C' linkage
 DIR *opendir(const char *);
      ^~~~~~~
In file included from /Users/thierry/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:39,
                 from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/thierry/.platformio/packages/framework-espidf/components/newlib/platform_include/sys/dirent.h:53:6: note: previous declaration with 'C++' linkage
 DIR* opendir(const char* name);
      ^~~~~~~
In file included from lib/Epub/EpubList/EpubIndex.h:13,
                 from src/main.cpp:9:
/Users/thierry/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/dirent.h:56:3: error: conflicting declaration of 'dirent* readdir(DIR*)' with 'C' linkage
```

sn0wst0rm, 2021-11-29:
That’s exactly my compile error. As @cgreening was saying it may be due to the different espidf toolchain downloaded with platformio.

@cgreening where did you get the `toolchain-xtensa32 2.80400.210211 (8.4.0) ` package? The version seems slightly different from the one provided with platformio

titimoby, 2021-11-29:
My version comes from PlateformIo extension, during an update

kvn-la, 2021-11-29:
I was having this same issue. I uninstalled Espressif 32 v3.4.0, and installed v3.3.0. This got rid of the errors seen in @sn0wst0rm's posts, and started to show the same versions as @cgreening. It did, however, fail with a new error - 

```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.3.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf 3.40300.0 (4.3.0) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - tool-ninja 1.9.0 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32 2.80400.210211 (8.4.0)
Reading CMake configuration...

*** Multiple ways to build the same target were specified for: /Users/klatzke/Code/esp32-ereader/.pio/build/m5_paper/bootloader/flash_encrypt.o  (from ['/Users/klatzke/.platformio/packages/framework-espidf/components/bootloader_support/src/flash_encrypt.c'] and from ['/Users/klatzke/.platformio/packages/framework-espidf/components/bootloader_support/src/esp32/flash_encrypt.c'])
File "/Users/klatzke/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 611, in compile_source_files
=========================================================================================================== [FAILED] Took 2.19 seconds ===========================================================================================================

Environment    Status    Duration
-------------  --------  ------------
m5_paper       FAILED    00:00:02.187
====================================================================================================== 1 failed, 0 succeeded in 00:00:02.187 ======================================================================================================
The terminal process "platformio 'run', '--target', 'upload', '--environment', 'm5_paper'" terminated with exit code: 1.
```

kvn-la, 2021-11-29:
Did a pull of the latest version of diy-esp32-epub-reader, that fixed the issue. 🤦‍♂️
It's now compiling as expected.

```
================================================================================================== [SUCCESS] Took 88.81 seconds ==================================================================================================

Environment    Status    Duration
-------------  --------  ------------
m5_paper       SUCCESS   00:01:28.808
```

titimoby, 2021-11-29:
@kvn-la the latest version compile but you need Espressif v3.3.0 ?

kvn-la, 2021-11-29:
@titimoby Correct. I had to have Espressif v3.3.0 installed.

cgreening, 2021-11-29:
Excellent - if you have any suggestions for how the docs can be improved
let me know and I’ll update the readme.

sn0wst0rm, 2021-11-29:
It seems a bug in the newer version of Espressif 32 Framework: https://github.com/espressif/esp-idf/issues/7204 and https://github.com/espressif/esp-idf/issues/7850.

They still did not released a patch as far as I know so I guess until then we'll have to stick with the older version

sn0wst0rm, 2021-11-29:
If you still want to use the latest version you can add a C linkage guard in file `/path/to/platformio/.platformio/packages/framework-espidf/components/newlib/platform_include/sys/dirent.h`:

```
// Copyright 2015-2016 Espressif Systems (Shanghai) PTE LTD
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

#pragma once

#include <stddef.h>
#include <stdint.h>

#ifdef __cplusplus
extern "C" {
#endif

/**
 * This header file provides POSIX-compatible definitions of directory
 * access functions and related data types.
 * See http://pubs.opengroup.org/onlinepubs/7908799/xsh/dirent.h.html
 * for reference.
 */

/**
 * @brief Opaque directory structure
 */
typedef struct {
    uint16_t dd_vfs_idx; /*!< VFS index, not to be used by applications */
    uint16_t dd_rsv;     /*!< field reserved for future extension */
    /* remaining fields are defined by VFS implementation */
} DIR;

/**
 * @brief Directory entry structure
 */
struct dirent {
    int d_ino;          /*!< file number */
    uint8_t d_type;     /*!< not defined in POSIX, but present in BSD and Linux */
#define DT_UNKNOWN  0
#define DT_REG      1
#define DT_DIR      2
#if __BSD_VISIBLE
#define MAXNAMLEN 255
    char d_name[MAXNAMLEN+1];   /*!< zero-terminated file name */
#else
    char d_name[256];
#endif
};

DIR* opendir(const char* name);
struct dirent* readdir(DIR* pdir);
long telldir(DIR* pdir);
void seekdir(DIR* pdir, long loc);
void rewinddir(DIR* pdir);
int closedir(DIR* pdir);
int readdir_r(DIR* pdir, struct dirent* entry, struct dirent** out_dirent);
#ifdef __cplusplus
}
#endif
```

kvn-la, 2021-11-30:
Has anyone else tried this, and has the m5paper worked afterwards? After building and uploading for me, I just get the hourglass, then a blank screen with the battery indicator in the corner. The buttons just bring up the hourglass again, but nothing else happens.

titimoby, 2021-11-30:
I need to find how to downgrade the Espressif version on Platformio first

sn0wst0rm, 2021-11-30:
@kvn-la i got the same result as you, I don’t know how to get it to work.
@titimoby you can edit the file I wrote about in my previous post, replacing it with the code I posted (or simply adding yourself the guard. If you want to use the older framework version you don’t need to downgrade, you need to download it from the embedded platform manager from platformio and then in the project config file you need to append `@3.3.0` to the framework config line.

cgreening, 2021-11-30:
@kvn-la Do you have an SD Card with some ePub files on it?

kvn-la, 2021-11-30:
I do, and I reinstalled the m5paper test program to ensure the card was being read. I only have one epub on there, so I'll put some more on and try it again in case it's something with the file. I'll check the SPIFFS option if that's not working.

cgreening, 2021-11-30:
Strange - anything in the logs? Let's open another issue as it sounds like something is going wrong somewhere,

kvn-la, 2021-11-30:
Actually, reformating the SD card and refilling it worked. Everything is working as expected now. Thanks for all your help!

cgreening, 2021-11-30:
That's great - there's a couple of bugs to track down in the content page (and some reformatting to make it a bit nicer).

closed 2021-11-30 by sn0wst0rm