atomic14/diy-esp32-epub-reader #75: Help in adding a language.

shojkeee, 2021-12-07:
Hello. Help please. I'm from Russia. And it turned out that this program does not support Cyrillic. Please add a language. Or please tell me how to add. I attach the book to the attachment for the test
[80072670.zip](https://github.com/atomic14/diy-esp32-epub-reader/files/7670231/80072670.zip)
.

martinberlin, 2021-12-08:
Hallo @shojkeee sure we will take a look.
@cgreening can you add here some steps to add a Cyrillic font?
Is there any other thing to consider when writing Cyrillic?

shojkeee, 2021-12-08:
> Hallo @shojkeee уверен , что мы будем смотреть. @cgreening, можете ли вы добавить сюда несколько шагов, чтобы добавить кириллический шрифт? Что еще нужно учитывать при написании кириллицы?

Here is a table of codes of Russian letters in utf8. Maybe this will help.
https://i.voenmeh.ru/kafi5/Kam.loc/inform/UTF-8.htm

Таблица кодов кириллицы в UTF-8
Символ	UNICODE	UTF-8	CP-1251
      Шестн.	Десят	Шестн.	Десят
А	0410	1040	D090	208 144	192
Б	0411	1041	D091	208 145	193
В	0412	1042	D092	208 146	194
Г	0413	1043	D093	208 147	195
Д	0414	1044	D094	208 148	196
Е	0415	1045	D095	208 149	197
Ж	0416	1046	D096	208 150	198
З	0417	1047	D097	208 151	199
И	0418	1048	D098	208 152	200
Й	0419	1049	D099	208 153	201
К	041A	1050	D09A	208 154	202
Л	041B	1051	D09B	208 155	203
М	041C	1052	D09C	208 156	204
Н	041D	1053	D09D	208 157	205
О	041E	1054	D09E	208 158	206
П	041F	1055	D09F	208 159	207
Р	0420	1056	D0A0	208 160	208
С	0421	1057	D0A1	208 161	209
Т	0422	1058	D0A2	208 162	210
У	0423	1059	D0A3	208 163	211
Ф	0424	1060	D0A4	208 164	212
Х	0425	1061	D0A5	208 165	213
Ц	0426	1062	D0A6	208 166	214
Ч	0427	1063	D0A7	208 167	215
Ш	0428	1064	D0A8	208 168	216
Щ	0429	1065	D0A9	208 169	217
Ъ	042A	1066	D0AA	208 170	218
Ы	042B	1067	D0AB	208 171	219
Ь	042C	1068	D0AC	208 172	220
Э	042D	1069	D0AD	208 173	221
Ю	042E	1070	D0AE	208 174	222
Я	042F	1071	D0AF	208 175	223
а	0430	1072	D0B0	208 176	224
б	0431	1073	D0B1	208 177	225
в	0432	1074	D0B2	208 178	226
г	0433	1075	D0B3	208 179	227
д	0434	1076	D0B4	208 180	228
е	0435	1077	D0B5	208 181	229
ж	0436	1078	D0B6	208 182	230
з	0437	1079	D0B7	208 183	231
и	0438	1080	D0B8	208 184	232
й	0439	1081	D0B9	208 185	233
к	043A	1082	D0BA	208 186	234
л	043B	1083	D0BB	208 187	235
м	043C	1084	D0BC	208 188	236
н	043D	1085	D0BD	208 189	237
о	043E	1086	D0BE	208 190	238
п	043F	1087	D0BF	208 191	239
р	0440	1088	D180	209 128	240
с	0441	1089	D181	209 129	241
т	0442	1090	D182	209 130	242
у	0443	1091	D183	209 131	243
ф	0444	1092	D184	209 132	244
х	0445	1093	D185	209 133	245
ц	0446	1094	D186	209 134	246
ч	0447	1095	D187	209 135	247
ш	0448	1096	D188	209 136	248
щ	0449	1097	D189	209 137	249
ъ	044A	1098	D18A	209 138	250
ы	044B	1099	D18B	209 139	251
ь	044C	1100	D18C	209 140	252
э	044D	1101	D18D	209 141	253
ю	044E	1102	D18E	209 142	254
я	044F	1103	D18F	209 143	255
Символы вне общего правила
Ё	0401	1025	D001	208 101	168
ё	0451	1025	D191	209 145	184

cgreening, 2021-12-08:
I think it's just a case of adding it to the font generation script. I'll do it abs document the steps.

cgreening, 2021-12-08:
The font intervals are defined in here: scripts/fontconvert.py so in theory we just add the ranges in and run the shell script... I'll try it and see what happens.

shojkeee, 2021-12-08:
> I think it's just a case of adding it to the font generation script. I'll do it abs document the steps.

Thanks. I will really wait for the result.

cgreening, 2021-12-08:
Will be later today/this evening.

martinberlin, 2021-12-08:
Hi Chris,
maybe it would be a nice feature if we can switch to this new font based on detecting language?
For example, the ePub provided as test has the following indicator showing is Russian in the **content.opf** XML:

    <dc:language>ru</dc:language>

Then if we detect **ru** language we can switch to use this new font. I think also [another languages](https://webgate.ec.europa.eu/sr/news/cyrillic-alphabet-day-2021) use Cyrillic apart of russian like:
Bulgarian, Macedonian and Serbian.

shojkeee, 2021-12-08:
Sorry. I started to understand the script myself fontconvert.py . Ready-made this script creates fonts well, but if you do not change it. The question is where to change the interval there. I can't understand it in any way. I see that the interval is changing in the script get_intervals_from_font.py . But this script is not involved in creating font files.

shojkeee, 2021-12-08:
"You can change which unicode character codes are to be exported by specifying additional ranges of unicode code points with --additional-intervals. Intervals are written as min,max. To add multiple intervals, you can specify the --additional-intervals option multiple times.

./fontconvert.py ... --additional-intervals 0xE0A0,0xE0A2 --additional-intervals 0xE0B0,0xE0B3 ..."

How to view the interval for Russian letters in hexadecimal format? I can't find :(

cgreening, 2021-12-08:
I've made the required changes here: https://github.com/atomic14/diy-esp32-epub-reader/pull/76

But - the current font doesn't have the code points defined here:https://en.wikipedia.org/wiki/List_of_Unicode_characters#Cyrillic

shojkeee, 2021-12-08:
Thank you very much. I'll try it now. But before that, can you please explain the following to me:
In the get_interval_from_font script.the interval is set to py. It is originally " for c in range(0, 65536):"

Where does this interval come from? I tried changing the dot interval from 0 to 209 145. And very few characters were added there. I haven't had time to check whether I succeeded or not. The question is how to find out the correct number of the interval for the letters I am interested in?

shojkeee, 2021-12-08:
More or less similar numbers to the truth are in the utf8 code table. Is that right? need a Utf8 code in decimal format?

cgreening, 2021-12-08:
I've updated the PR to use open sans as the fallback. 

In the script folder there are two files `fontconvert.py` and `generate_fonts.sh`. I just add the new codepoints to the `fontconvert.py` and then run the `generate_fonts.sh`.

cgreening, 2021-12-08:
It's the code points in hexidecimal UTF-16. For now I've taken the code block from here: https://en.wikipedia.org/wiki/List_of_Unicode_characters#Cyrillic which is:

```
    (0x400, 0x486),  # https://en.wikipedia.org/wiki/List_of_Unicode_characters#Cyrillic
    (0x489, 0x4FF),  # https://en.wikipedia.org/wiki/List_of_Unicode_characters#Cyrillic
```

There's a couple of missing characters that I've ignore for now.

shojkeee, 2021-12-08:
> I've updated the PR to use open sans as the fallback.
> 
> In the script folder there are two files `fontconvert.py` and `generate_fonts.sh`. I just add the new codepoints to the `fontconvert.py` and then run the `generate_fonts.sh`.

I saw that, just generate_font.sh generates 4 files at once and throws them into the lib/font folder. And I ran commands from this file separately to see what it creates.

cgreening, 2021-12-08:
Try the code in this branch and see if it works - `feature/add-cyrillic`

shojkeee, 2021-12-08:
Excuse me. And you can't give a link to this branch to download it. I don't know much about github. Yesterday, when I tried to build the program, I got an error until I downloaded it with the git clone --recursive command https://github.com/atomic14/diy-esp32-epub-reader.git

I think the link should be the same approximately?

martinberlin, 2021-12-08:
Hello @shojkeee 
please checkout the branch: **feature/add-cyrillic**

Set the platformio.ini to: default_envs = m5_paper
And build again. It builds correctly for me:
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   9.8% (used 32048 bytes from 327680 bytes)
Flash: [========= ]  91.4% (used 1078741 bytes from 1179648 bytes)

Environment    Status    
-------------  --------  
m5_paper       SUCCESS

shojkeee, 2021-12-08:
> Hello @shojkeee please checkout the branch: **feature/add-cyrillic**
>
> Set the platformio.ini to: default_envs = m5_paper And build again. It builds correctly for me: Advanced Memory Usage is [...]

Please help me. What command was entered to download the project from this branch? If I 'm pumping .zip archive, then does not compile at all swears at libraries.

cgreening, 2021-12-08:
Git can be pretty confusing if you've not used it much.

```
git clone --recursive git@github.com:atomic14/diy-esp32-epub-reader.git
cd diy-esp32-epub-reader
git checkout feature/add-cyrillic
```

The above will clone the repository with all the submodules into `diy-esp32-epub-reader`. And then you change into that directory and checkout the branch.

shojkeee, 2021-12-08:
@cgreening  Thanks for the help with git. Thank you for adding Russian. Now I checked on a book that was not read. Everything is working perfectly now. Will you add Russian to the main branch? I want to make a video about it, I have 15k subscribers on YouTube, I think they will need it.

[image]

martinberlin, 2021-12-08:
Tested here and also works
[image]
[image]
@cgreening I think is ready to be merged reviewing it now.

cgreening, 2021-12-08:
I wonder if we should just use opensans as our default font?

cgreening, 2021-12-08:
@shojkeee Yes! Please do a video - that would be great - once we've merged the pull request it will be in the main branch.

cgreening, 2021-12-08:
All done.

closed 2021-12-08 by cgreening