atomic14/diy-esp32-epub-reader #78: LilyGo_T5_47 upload the code enter reboot cycle

trumeen, 2022-01-03:
After upload the code lilygo enter reboot cycle. The log is below.

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4996
load:0x40078000,len:14436
load:0x40080400,len:3240
entry 0x40080610
I (481) psram: This chip is ESP32-D0WD
I (482) spiram: Found 64MBit SPI RAM device
I (482) spiram: SPI RAM mode: flash 80m sram 80m
I (484) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (492) cpu_start: Pro cpu up.
I (495) cpu_start: Starting app cpu, entry point is 0x400840c0
I (489) cpu_start: App cpu up.
I (517) cpu_start: Pro cpu start user code
I (517) cpu_start: cpu freq: 240000000
I (517) cpu_start: Application information:
I (522) cpu_start: Project name:     epub-reader
I (527) cpu_start: App version:      e849c1e-dirty
I (533) cpu_start: Compile time:     Jan  3 2022 18:07:48
I (539) cpu_start: ELF file SHA256:  bf1bf981faa088a8...
I (545) cpu_start: ESP-IDF:          4.3.1
I (550) heap_init: Initializing. RAM available for dynamic allocation:
I (557) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (563) heap_init: At 3FFB80E0 len 00027F20 (159 KiB): DRAM
I (569) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (575) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (582) heap_init: At 40095C2C len 0000A3D4 (40 KiB): IRAM
I (588) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (596) spi_flash: detected chip: generic
I (600) spi_flash: flash io: dio
I (610) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (615) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (625) main: epub list state num_epubs=0
I (625) main: epub list state is_loaded=0
I (635) main: epub list state selected_item=0
I (635) main: Memory before main task start 4413491
I (645) main: Powering up the board
I (645) main: Creating renderer
I (715) main: Starting file system
I (715) main: Using SPIFFS
I (715) SPIFFS: Initializing SPIFFS
I (935) main: Starting battery monitor
I (935) main: Setting up controls
i2c_driver started correctly
I (935) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:1 
E (945) gpio: gpio_install_isr_service(460): GPIO isr service already installed
ISR trigger install response: 0x103
I (955) main: Controls configured
I (955) EPD: Full clear

abort() was called at PC 0x400851c7 on core 0

Backtrace:0x4008ed8e:0x3ffbe860 0x4008f509:0x3ffbe880 0x40094862:0x3ffbe8a0 0x400851c7:0x3ffbe910 0x40085325:0x3ffbe940 0x400853c1:0x3ffbe960 0x4012eaed:0x3ffbe990 0x40125f13:0x3ffbec50 0x40125d36:0x3ffbeca0 0x40092307:0x3ffbecd0 0x4008fdcf:0x3ffbed00 0x4008307d:0x3ffbed40 0x4009213d:0x3ffbf170
  #0  0x4008ed8e:0x3ffbe860 in panic_abort at C:\Users\ddj1\.platformio\packages\framework-espidf\components\esp_system/panic.c:368        
  #1  0x4008f509:0x3ffbe880 in esp_system_abort at C:\Users\ddj1\.platformio\packages\framework-espidf\components\esp_system/system_api.c:112
  #2  0x40094862:0x3ffbe8a0 in abort at C:\Users\ddj1\.platformio\packages\framework-espidf\components\newlib/abort.c:46
  #3  0x400851c7:0x3ffbe910 in lock_acquire_generic at C:\Users\ddj1\.platformio\packages\framework-espidf\components\newlib/locks.c:139   
  #4  0x40085325:0x3ffbe940 in _lock_acquire_recursive at C:\Users\ddj1\.platformio\packages\framework-espidf\components\newlib/locks.c:167  #5  0x400853c1:0x3ffbe960 in __retarget_lock_acquire_recursive at C:\Users\ddj1\.platformio\packages\framework-espidf\components\newlib/locks.c:323
  #6  0x4012eaed:0x3ffbe990 in _vfiprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:853 (discriminator 2)
  #7  0x40125f13:0x3ffbec50 in fiprintf at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fiprintf.c:48
  #8  0x40125d36:0x3ffbeca0 in __assert_func at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:58 (discriminator 8)
  #9  0x40092307:0x3ffbecd0 in spinlock_acquire at C:\Users\ddj1\.platformio\packages\framework-espidf\components\esp_hw_support\include/soc/spinlock.h:122
      (inlined by) vPortCPUAcquireMutex at C:\Users\ddj1\.platformio\packages\framework-espidf\components\freertos\port\xtensa\include/freertos/portmacro.h:158
      (inlined by) vPortEnterCritical at C:\Users\ddj1\.platformio\packages\framework-espidf\components\freertos\port\xtensa/port.c:448    
  #10 0x4008fdcf:0x3ffbed00 in xQueueGenericSend at C:\Users\ddj1\.platformio\packages\framework-espidf\components\freertos/queue.c:790    
  #11 0x4008307d:0x3ffbed40 in provide_out at lib/epdiy/lut.c:427
  #12 0x4009213d:0x3ffbf170 in vPortTaskWrapper at C:\Users\ddj1\.platformio\packages\framework-espidf\components\freertos\port\xtensa/port.c:168

ELF file SHA256: bf1bf981faa088a8

cgreening, 2022-01-04:
I'll take a look - can you confirm that you uploaded the filesystem as well as the code?

cgreening, 2022-01-04:
I can recreate this - I'll see if something has changed in the epd library.

cgreening, 2022-01-04:
Actually - I get the watchdog timeout, but it can be safely ignored. Can you check that you have uploaded the SPIFFs filesystem with some books in it?

trumeen, 2022-01-05:
I used the "pio run -t uploadfs", but the problem still exists. The log is below. Could you help me? TKS

Building in release mode
Building SPIFFS image from 'data' directory to .pio\build\lilygo_t5_47\spiffs.bin
/entities_ww2.epub
/pg43-images.epub
Looking for upload port...
Use manually specified: COM3
Uploading .pio\build\lilygo_t5_47\spiffs.bin
esptool.py v3.1
Serial port COM3
Connecting........___
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 34:ab:95:5d:10:1c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 16MB
Flash will be erased from 0x00130000 to 0x003fffff...
………………………………………………………………………………

trumeen, 2022-01-05:
Sometimes the log is different。

Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40093bc0  PS      : 0x00060c33  A0      : 0x80094542  A1      : 0x3ffd0930
A2      : 0x3f800014  A3      : 0x00000017  A4      : 0x3ffb626c  A5      : 0x00060c20  
A6      : 0x000000fe  A7      : 0x00000001  A8      : 0x3f847540  A9      : 0x3f8869c4
A10     : 0x0000000f  A11     : 0x1f8840f5  A12     : 0x0037962f  A13     : 0x3f8869cc  
A14     : 0x3f847548  A15     : 0xfffffffc  SAR     : 0x00000010  EXCCAUSE: 0x0000001d
EXCVADDR: 0x1f884101  LBEG    : 0x40089d64  LEND    : 0x40089d80  LCOUNT  : 0xffffffff  

Backtrace:0x40093bbd:0x3ffd0930 0x4009453f:0x3ffd0950 0x40084ee5:0x3ffd0970 0x400da78f:0x3ffd0990 0x400d339b:0x3ffd09c0 0x400d4142:0x3ffd0a00 0x4009213d:0x3ffd0a60
  #0  0x40093bbd:0x3ffd0930 in insert_free_block at C:\Users\ddj1\.platformio\packages\framework-espidf\components\heap/heap_tlsf.c:241      (inlined by) block_insert at C:\Users\ddj1\.platformio\packages\framework-espidf\components\heap/heap_tlsf.c:269
      (inlined by) block_trim_free at C:\Users\ddj1\.platformio\packages\framework-espidf\components\heap/heap_tlsf.c:383
      (inlined by) block_prepare_used at C:\Users\ddj1\.platformio\packages\framework-espidf\components\heap/heap_tlsf.c:460
      (inlined by) tlsf_malloc at C:\Users\ddj1\.platformio\packages\framework-espidf\components\heap/heap_tlsf.c:758
  #1  0x4009453f:0x3ffd0950 in multi_heap_malloc_impl at C:\Users\ddj1\.platformio\packages\framework-espidf\components\heap/multi_heap.c:197
  #2  0x40084ee5:0x3ffd0970 in heap_caps_malloc at C:\Users\ddj1\.platformio\packages\framework-espidf\components\heap/heap_caps.c:145 
  #3  0x400da78f:0x3ffd0990 in epd_hl_init at lib/epdiy/highlevel.c:35
  #4  0x400d339b:0x3ffd09c0 in EpdiyRenderer::EpdiyRenderer(EpdFont const*, EpdFont const*, EpdFont const*, EpdFont const*, unsigned char const*, int, int) at lib/Epub/Renderer/EpdiyRenderer.h:28
      (inlined by) Lilygo_t5_47::get_renderer() at src/boards/Lilygo_t5_47.cpp:38
  #5  0x400d4142:0x3ffd0a00 in main_task(void*) at src/main.cpp:207
  #6  0x4009213d:0x3ffd0a60 in vPortTaskWrapper at C:\Users\ddj1\.platformio\packages\framework-espidf\components\freertos\port\xtensa/port.c:168

ELF file SHA256: 4f471b1563bc115a

trumeen, 2022-01-17:
If I use the battery ,the problem does not exist. So I closed the issue.

closed 2022-01-17 by trumeen