atomic14/diy-esp32-epub-reader #79: Continuous hour glas

pcmdx, 2022-02-25:
I loaded the project via M5Burner and put the recommended epub file on the micro SD card. However, despite restarts/etc all I get is the hour glass, even after waiting a while. Is this user error, in which case can you provide a pointer? Thanks.

codingjoe, 2022-02-27:
@pcmdx same here. Not all EPUB files seem to work well. Honestly, EPUB isn't the best format for an esp32 to being with, due to its limited capabilities. If you want to use your P5Paper as an e-reader, I'd recommend preprocessing the EPUBs. A text only version should work mostly, but then you could have also just written a small app that reads plain text files.
I recently wrote my own bitmap renderer for the M5Paper, since the performance to render compressed formats was poor. Since most EPUBs will use JPEG, you'll have similar issues, I presume.

pcmdx, 2022-02-27:
@codingjoe Some good points. With UI Flow now available I might try MicroPython with text files.

codingjoe, 2022-02-28:
At least the Gutenberg Project has all books as plain text. If you develop one, I'd love to see one with a beautiful type face, like Vollkorn.

cgreening, 2022-02-28:
Can you post any logs coming from the device? That will help troubleshoot the issue.

Some ePub files are just too big for the esp32 to handle. But most should work.

If you want the best results then I would suggest rendering the ePub files to images and then have a simple slideshow.

pcmdx, 2022-02-28:
Please find screenshot below @cgreening. It seems to be a file open failed with the file the M5Burner tool recommends:

https://m5stack-firmware.oss-cn-shenzhen.aliyuncs.com/files/pg14838-images.epub

I renamed it to zip on my local machine and it unzips and loads the container.xml file successfully.

[image]

cgreening, 2022-02-28:
Can you attach the ePub file you are using and I'll give it a go. We can also add some more robust error handling so it doesn't fail quite so badly when it can read the ePub contents.

pcmdx, 2022-02-28:
Thanks @cgreening. Please find attached.

pcmdx, 2022-02-28:
Zipped copy. [pg14838-images epub.zip](https://github.com/atomic14/diy-esp32-epub-reader/files/8153436/pg14838-images.epub.zip)

cgreening, 2022-02-28:
Great - will try it tonight.

cgreening, 2022-02-28:
I've tried with the ePub you sent and it seems to work. I'm using PlatformIO and not the M5Burner - I'm not sure how the M5Burner works. Can you try using PlatformIO instead?

Can you make sure that your M5Stack is fully charged up - I have a lot of problems using the SDCard when the device does not have much power - it immediately crashes.

Can you try using the SPIFFS filesystem option to see if that works.

In `platformio.ini` change line 150 so that it is uncommented - that will switch to using SPIFFS instead of the SDCard.

You will need to "upload the filesystem" using platform io.

[image]

I know there's nothing worse than someone saying it, but it works on my machine...

[image]

pcmdx, 2022-03-01:
Thanks for the detailed feedback and testing @cgreening. I did a build and upload using PlatformIO, which went smoothly, however, it doesn't seem to be loading the program at all now. When I monitor the serial port I get:

```
rst:0x1 (POWERON_RESET),boot:0x4 (DOWNLOAD_BOOT(UART0/UART1/SDIO_FEI_FEO_V2))
waiting for download
```

Here's the full upload process, which looks like it worked OK. Am I missing a boot loader or some other preliminary step not carried out by the build?

Thanks again for your help.

```
> Executing task: C:\Users\paulm\.platformio\penv\Scripts\platformio.exe run --target uploadfs --environment m5_paper <

Processing m5_paper (platform: espressif32; board: esp32dev; framework: espidf)
-----------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.5.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf 3.40302.0 (4.3.2) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - tool-idf 1.0.1 
 - tool-mconf 1.4060000.20190628 (406.0.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - tool-ninja 1.9.0 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-riscv32-esp 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32s2 8.4.0+2021r2-patch2
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <tinyxml2> 0.0.0+20220301064642.sha.a977397
|-- <sd_card>
|-- <spiffs>
|-- <Fonts>
|-- <Images>
|-- <Epub>
|   |-- <miniz-2.2.0>
|   |-- <tjpgd3>
|   |-- <m5Paper>
|   |-- <tinyxml2> 0.0.0+20220301064642.sha.a977397
Building in release mode
Building SPIFFS image from 'data' directory to .pio\build\m5_paper\spiffs.bin
/.keep
/pg14838-images.epub
/pg43-images.epub
Looking for upload port...
Auto-detected: COM4
Uploading .pio\build\m5_paper\spiffs.bin
esptool.py v3.1
Serial port COM4
Connecting....
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:3a:f2:66:d6:6c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 16MB
Flash will be erased from 0x00130000 to 0x003fffff...
Compressed 2949120 bytes to 1742771...
Writing at 0x00130000... (0 %)
Writing at 0x001366f1... (1 %)
Writing at 0x0013c5e9... (2 %)
REDACTED
Writing at 0x003ee51d... (98 %)
Writing at 0x003f55c0... (99 %)
Writing at 0x003fd4e9... (100 %)
Wrote 2949120 bytes (1742771 compressed) at 0x00130000 in 39.2 seconds (effective 602.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
======================================================== [SUCCESS] Took 52.66 seconds ========================================================

Environment    Status    Duration
-------------  --------  ------------
m5_paper       SUCCESS   00:00:52.655
========================================================= 1 succeeded in 00:00:52.655 =========================================================
```

cgreening, 2022-03-01:
Can you follow the upload file system with a normal upload. Should be in the group above the one I highlighted.

pcmdx, 2022-03-01:
If I try m5-Paper -> General -> Upload, I get the following output, which looks like it's missing source for a driver.

```
> Executing task: C:\Users\paulm\.platformio\penv\Scripts\platformio.exe run --target upload --environment m5_paper <

Processing m5_paper (platform: espressif32; board: esp32dev; framework: espidf)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.5.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf 3.40302.0 (4.3.2) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - tool-idf 1.0.1 
 - tool-mconf 1.4060000.20190628 (406.0.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - tool-ninja 1.9.0 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-riscv32-esp 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32s2 8.4.0+2021r2-patch2
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <tinyxml2> 0.0.0+20220301064642.sha.a977397
|-- <sd_card>
|-- <spiffs>
|-- <Fonts>
|-- <Images>
|-- <Epub>
|   |-- <miniz-2.2.0>
|   |-- <tjpgd3>
|   |-- <m5Paper>
|   |-- <tinyxml2> 0.0.0+20220301064642.sha.a977397
Building in release mode
Generating ULP assembly file .pio\build\m5_paper\ulp_main.bin.S
Compiling .pio\build\m5_paper\src\boards\Epdiy.o
Compiling .pio\build\m5_paper\src\boards\Lilygo_t5_47.o
Compiling .pio\build\m5_paper\src\boards\M5Paper.o
Compiling .pio\build\m5_paper\src\boards\controls\EpdiyV6ButtonControls.o
In file included from src/boards/controls/EpdiyV6ButtonControls.cpp:1:
src/boards/controls/EpdiyV6ButtonControls.h:2:10: fatal error: pca9555.h: No such file or directory

*****************************************************************
* Looking for pca9555.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pca9555.h"
* Web  > https://registry.platformio.org/search?q=header:pca9555.h
*
*****************************************************************

 #include "pca9555.h"
          ^~~~~~~~~~~
compilation terminated.
*** [.pio\build\m5_paper\src\boards\controls\EpdiyV6ButtonControls.o] Error 1
In file included from src/boards/Epdiy.cpp:2:
lib/Epub/Renderer/EpdiyRenderer.h:3:10: fatal error: epd_driver.h: No such file or directory

********************************************************************
* Looking for epd_driver.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:epd_driver.h"
* Web  > https://registry.platformio.org/search?q=header:epd_driver.h
*
********************************************************************

 #include <epd_driver.h>
          ^~~~~~~~~~~~~~
compilation terminated.
In file included from src/boards/Lilygo_t5_47.cpp:2:
lib/Fonts/regular_font.h:2:10: fatal error: epd_driver.h: No such file or directory

********************************************************************
* Looking for epd_driver.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:epd_driver.h"
* Web  > https://registry.platformio.org/search?q=header:epd_driver.h
*
********************************************************************

 #include "epd_driver.h"
          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\m5_paper\src\boards\Epdiy.o] Error 1
*** [.pio\build\m5_paper\src\boards\Lilygo_t5_47.o] Error 1
In file included from lib/Epub/Renderer/M5PaperRenderer.h:6,
                 from src/boards/M5Paper.cpp:2:
lib/Epub/Renderer/EpdiyFrameBufferRenderer.h:3:10: fatal error: epd_driver.h: No such file or directory

********************************************************************
* Looking for epd_driver.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:epd_driver.h"
* Web  > https://registry.platformio.org/search?q=header:epd_driver.h
*
********************************************************************

 #include <epd_driver.h>
          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\m5_paper\src\boards\M5Paper.o] Error 1
================================================================================ [FAILED] Took 9.00 seconds ================================================================================

Environment    Status    Duration
-------------  --------  ------------
m5_paper       FAILED    00:00:09.001
=========================================================================== 1 failed, 0 succeeded in 00:00:09.001 ===========================================================================
The terminal process "C:\Users\paulm\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload', '--environment', 'm5_paper'" terminated with exit code: 1.
```

cgreening, 2022-03-01:
I think when you cloned the repo you didn't clone it recursively so it doesn't have any of the submodules.

Not a problem, you can either try:

```
git submodule update --recursive
```

Or just do a fresh clone with:

```
git clone --recursive git@github.com:atomic14/esp32-ereader.git
```

pcmdx, 2022-03-01:
Ah, what a stupid mistake I made. You're right @cgreening, and that now works perfectly!

For context on my original request, this is what the M5Burner interface looks like. I'm not sure if someone needs contacting, but they're clearly promoting a download that doesn't work through that tool.

Thanks again for your help.

[image]

cgreening, 2022-03-01:
No problem - I'll see if I can find a contact there.

closed 2022-03-01 by cgreening