atomic14/diy-esp32-epub-reader #80: Error reading files from sdcard on EPD47

TheBattleWolf, 2022-03-27:
Hello,

I was tying to use the official [LILYGO TF-Card adapter](https://de.aliexpress.com/item/1005002317501092.html) to get microSD working on the [EPD47](https://de.aliexpress.com/item/1005002006058892.html). But reading any files from the FAT32 formatted card results in the following error:
```
I (702) SDC: Initializing SD card
I (702) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (732) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
I (732) gpio: GPIO[12]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (732) SDC: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.
I (752) main: Starting battery monitor
I (752) main: Setting up controls
I (752) main: Controls configured
I (762) EPD: Full clear
I (2132) main: Creating epub list
No such file or directory
91D79:0x3FFC0620 

9255:0x3FFC05E0 0x400906F9:0x3FFC0600 0x400
ingE (5722) task_wdt:  - IDLE (CPU 1)
E (5722) task_wdt: Tasks currently running:
E (5722) task_wdt: CPU 0: epd_fetch
E (5722) task_wdt: CPU 1: IDLE
E (5722) task_wdt: Print CPU 0 (current core) backt91D79:0x3FFC0620 

E (5922) PUBLIST: Is SD-Card inserted and properly connected?
Could not open directory /fs/
```
These are my changes to the platformio file:
```
-  -D USE_SPIFFS
+  ;-D USE_SPIFFS
   
   ; Logging. Leave enabled for first builds and debugging. Comment to disable
   -D LOG_ENABLED
   ; Touch configuration
-  -D USE_L58_TOUCH
-  -D CONFIG_TOUCH_SDA=15
-  -D CONFIG_TOUCH_SDL=14
-  -D CONFIG_TOUCH_INT=13
+  ;-D USE_L58_TOUCH
+  -D CONFIG_TOUCH_SDA=0
+  -D CONFIG_TOUCH_SDL=0
+  -D CONFIG_TOUCH_INT=0
   -D CONFIG_I2C_MASTER_FREQUENCY=50000
   -D CONFIG_FT6X36_DEBUG=0
+lib_ignore = 
+  touch
+  FT6X36
```

I've tried SPIFFS and it works fine. I've also tried lowering m_host.max_freq_khz to 5000 but that also didn't change anything. Is there anything obvious I am missing here?

Any help is greatly appreciated :-). Let me know if you need any more details.

closed 2022-05-12 by TheBattleWolf

TheBattleWolf, 2022-11-29:
Seems like the Problem was caused by the adapter not beeing connected entirely. The issue was solved using a pair of pliers to connect it all the way.

RWBRWB, 2022-12-28:
@TheBattleWolf I'm needing to add a SD card to the EDP4.7 also but I can't find any code that shows how to write images that are stored on the SD card to display. 

Can you let me know where you found this code or provide an example that shows how it works? 

Any help is appreciated!

TheBattleWolf, 2022-12-28:
> @TheBattleWolf I'm needing to add a SD card to the EDP4.7 also but I can't find any code that shows how to write images that are stored on the SD card to display.
> 
> Can you let me know where you found this code or provide an example that shows how it works?
> 
> Any help is appreciated!

I'm not sure what the issue is right now. Maybe you can clarify your question. The "images" should be stored in the epub file. The code to draw them is located in [lib/Epub/Renderer](https://github.com/atomic14/diy-esp32-epub-reader/tree/main/lib/Epub/Renderer). If you are having trouble displaying any ebook, try applying the diff above and upload the code. After that copy the example ebook onto the fat32 formatted sdcard(not all epubs work).