atomic14/diy-esp32-epub-reader #81: Weird graphical glitch in the display of content.

rttgnck, 2022-04-03:
I'm having a weird text issue, but it happens on the battery icon and the book selection screen as well. Everything had this weird shift in the display. Any idea what it might be?
[image]
[image]

dronix27, 2022-08-23:
The same in Cyrillic

rttgnck, 2022-08-23:
I think I had solved it sometime back and forgot to say something. I'll try to dig back into my code and see what I did or if I left myself any clues

dronix27, 2022-08-25:
> I think I had solved it sometime back and forgot to say something. I'll try to dig back into my code and see what I did or if I left myself any clues

I really hope you found a solution and share it

rttgnck, 2022-08-25:
> I really hope you found a solution and share it

So I will have to look further, possibly fork, and upload my changes. 

At first glance, it looks like I rendered a new font.
I also changed a section of the epd_driver.c which is part of the epdiy repo, not this one.
Pretty sure I also changed the font size, or line height, to get a few more words on a page.

In my personal changes I was adding a section page count, and trying to add a bookmark of sorts so when the unit restarts you can jump back to the page you were on (not even close to have that).

It would be a little while before I had a chance to clean up my local copy for a fork.

dronix27, 2022-09-08:
Is the problem still not resolved?

rttgnck, 2022-09-08:
I have not explored anything further. It's unfortunately not really top of
my priority list right now.

Lefucjusz, 2023-11-16:
I've run into the exact same problem on my M5Paper, it looks that the nibbles in each byte sent to the display are swapped. I didn't read very deeply into the code, but it looks that swapping them in `epd_driver.c:epd_draw_pixel()` resolves the issue. The original code looks like this:

```
void epd_draw_pixel(int x, int y, uint8_t color, uint8_t *framebuffer) {

  ...some stuff here...

  uint8_t *buf_ptr = &framebuffer[y * EPD_WIDTH / 2 + x / 2];
  if (x % 2) {
    *buf_ptr = (*buf_ptr & 0x0F) | (color & 0xF0);
  } else {
    *buf_ptr = (*buf_ptr & 0xF0) | (color >> 4);
  }
}
```

Just change the condition to `if (!(x % 2))` or exchange the lines between `if` and `else` and the image will no longer be weirdly shifted.