atomic14/diy-esp32-epub-reader #86: EPD47 not booting with microSD connected

TheBattleWolf, 2022-11-29:
Hello,

I have successfully used the official [LILYGO TF-Card adapter](https://de.aliexpress.com/item/1005002317501092.html) on my [EPD47](https://de.aliexpress.com/item/1005002006058892.html) to get access to the Files on my microSD card. But if I power on the device while the adapter is inserted the device doesn't boot (even if the card is not in the adapter). I only got it to work if I connect the adapter after the device is booted, at the next auto reboot the books are loaded. To narrow down the problem I flashed the [LilyGo demo](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/tree/master/examples/demo) via Arduino IDE. After flashing the demo the device starts fine with the adapter and card inserted, the card is recognized and the size of the card is displayed.

Any help fixing the problem is greatly appreciated.

Thanks in advance.

cgreening, 2022-11-29:
That's interesting, any logs from the initial boot where it fails?

TheBattleWolf, 2022-11-29:
> That's interesting, any logs from the initial boot where it fails?

Yes, this is the output that loops.
```
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:5388
ho 0 tail 12 room 4
load:0x40078000,len:15012
load:0x40080400,len:3220
entry 0x4008060c
I (494) psram: This chip is ESP32-D0WD
E (495) psram: PSRAM ID read error: 0xfffbff0f
E (495) cpu_start: Failed to init external RAM!
Re-enable cpu cache.

abort() was called at PC 0x400849a1 on core 0

Backtrace:0x40084d3a:0x3ffe3bb0 0x4008f551:0x3ffe3bd0 0x40095132:0x3ffe3bf0 0x400849a1:0x3ffe3c60 0x40079543:0x3ffe3c90 |<-CORRUPTED
  #0  0x40084d3a:0x3ffe3bb0 in panic_abort at /home/thebattlewolf/.platformio/packages/framework-espidf/components/esp_system/panic.c:393
  #1  0x4008f551:0x3ffe3bd0 in esp_system_abort at /home/thebattlewolf/.platformio/packages/framework-espidf/components/esp_system/system_api.c:112
  #2  0x40095132:0x3ffe3bf0 in abort at /home/thebattlewolf/.platformio/packages/framework-espidf/components/newlib/abort.c:46
  #3  0x400849a1:0x3ffe3c60 in call_start_cpu0 at /home/thebattlewolf/.platformio/packages/framework-espidf/components/esp_system/port/cpu_start.c:360 (discriminator 1)
  #4  0x40079543:0x3ffe3c90 in ?? ??:0

ELF file SHA256: 0a6415f14a2ce73c

Rebooting...
```

BernhardtL, 2024-01-16:
Not shure if it helps, but I struggled for some time with the same and other problemes.

1. Check GPIO Pin Numbers in PlatformIO.ini with those in the Lilygo Demo
2. Disable SPDIFF Storage as it isnt working with the SD Card functions
3. Disable Touch, as far as I understand the Issue, some Lilygo Versions wont work with SD and touch

Long answer:

1. As way to often the case, the Lilygo documentation is all over the place and newer boards replace older boards faster than the docs update. In my case I use the Version with the ESP32 WROVER (not the S3!), this Version is only able to read the SD card with following changes:
```
;  ; setup the pins for the SDCard
 ;-DSD_CARD_PIN_NUM_MISO=GPIO_NUM_14
 ;-DSD_CARD_PIN_NUM_MOSI=GPIO_NUM_13
 ;-DSD_CARD_PIN_NUM_CLK=GPIO_NUM_15
 ;-DSD_CARD_PIN_NUM_CS=GPIO_NUM_12
    ; setup the pins for the SDCard for EPD47 WROVER
  -DSD_CARD_PIN_NUM_MISO=GPIO_NUM_12
  -DSD_CARD_PIN_NUM_MOSI=GPIO_NUM_13
  -DSD_CARD_PIN_NUM_CLK=GPIO_NUM_14
  -DSD_CARD_PIN_NUM_CS=GPIO_NUM_15
```
This change must be also (but to different GPIOs) done with certain other EPD47 Versions, as mentioned above, test with the Lilygo Demo and copy the Pin Numbers.

2. It took me some time to comprehend, that the internal memory function is only implemented as a rough prototype. For full features, like saving the page when switching books is only supported with the SD. Though it was fun to test SPIFFS.
Deactivate it correctly like this:
```
  ; use SPIFFS - experimental feature
  ;-D USE_SPIFFS 
```
3. In my version the touch pins are also the SD pins, so only either or works. Maybe it´s possible to use other touch pins, but I prefer buttons over touch anyway. It took me some time to figure out, how to disable touch. First I tried deleting everything touch related, but it turns out there is an easier way, almost certainly the intended way:
```
  ; Touch configuration
  ;-D USE_L58_TOUCH
  -D CONFIG_TOUCH_SDA=15
  -D CONFIG_TOUCH_SDL=14
  -D CONFIG_TOUCH_INT=13
```

Still the inserted SD Card sometimes troubles somehow the reset, but I think this is more related to the way, Lilygo build the Reset Function/Button. As it sometimes enters the flash/upload mode and not a reset.

TheBattleWolf, 2024-01-17:
Hey, as I wrote in the initial issue, accessing the files is not the problem, see my initial comment on this issue. I also have touch and spiffs disabled and the GPIO pins changed. I also have the WROVER version, are you using the official sdcard adapter?