atomic14/diy-esp32-epub-reader #88: Bug: M5Paper battery drain

klundry, 2023-01-09:
I'm not sure if this is a bug with the firmware or just a quirk of the M5Paper itself, but the battery seems to drain rather fast when it is "deep sleeping". More than it should in my view. For example, starting from fully charged and leaving it untouched for a day the battery will be more than half drained when woken back up. If I charge it then press the power off button on the back and let it sit it won't drain at all.

If this is just a quirk of the M5Paper design that can't be worked around then the save progress/bookmark feature I mentioned in #87 plus pressing the button on the back would be a potential workaround.

martinberlin, 2023-02-06:
Hello Chris,
Will you still give some support for this Firmware ?
I don't own one of this models but I'm aware that IT8951 has a nice battery consumption so that might be well the case.

cgreening, 2023-02-07:
Sounds like an issue with the M5Paper - maybe it's not switching off the ePaper power supply when going into deep sleep?

martinberlin, 2023-02-07:
It might be. IT8951 when it get's the deepsleep command via SPI consumes minimally.
But in my own PCB design to control it I'm directly pulling off the plug:
https://github.com/martinberlin/H-cinread-it8951/tree/main/Schematic

If you can try to check how much it consumes the sleeping M5. It will be nice to know at least.