atomic14/diy-esp32-epub-reader #89: problems during building

lionlggnd, 2023-01-26:
i can't build the project on studio code
i clone git but every time i try to build the project the command line comes up with errors of some kind.
Steps:
1) I open SC
2) I click on the platformio menu and open the home screen
3) select open folder on homepage, select the folder downloaded from git
4) project opens, after that i open the platformio icon again, click lilygo - build
5) numerous errors come out during the construction phase, especially on the non-recognition of dependencies and #include tags.

How can I do?
Alternatively, can you provide me the pre-compiled binaries for lilygo?
Thank you.

martinberlin, 2023-02-03:
Are you selecting the right enviroment in the **platformio.ini** file?
Select the lilygo_t5_47 default option and try again. It builds correctly for me. 

Maybe you have an older version of Espressif platform?
Try to update that and try again.

lionlggnd, 2023-02-04:
https://user-images.githubusercontent.com/117315249/216768747-87184714-f626-43b0-99cc-f1f888f8395a.mp4

martinberlin, 2023-02-06:
Wait!
But you need to run: 

    pio run -t menuconfig

In a terminal there in Platformio and setup the Display and board in components > epdiy
Otherwise how your Firmware will be aware of what display /board you are using?

in-fluxed, 2023-02-07:
I had the same problem when I downloaded project as a zip. If I used git it builds successfully.