atomic14/diy-esp32-epub-reader #91: HELP

FrancescSobrevela, 2023-02-11:
I'm new on this world, and I need some help to upload the ebook reader to my lilygo T5 47 S3. Someone can explain how can I do it? 
Thanks

MarcelvLaar, 2023-02-12:
Watch [this](https://youtu.be/VLiCgB0odOQ) video, and read the README completely. your Lilygo model is listed under supported, so i assume if you just follow the instructions (build and flash this program with platformIO) It will perform as expected. 

general things like building and flashing to an ESP32 or arduino you can just google. If follow the readme / video, and you do it right, it'll probably work.

Also, please don't put a general "explain everything please" in bold letters in the issues page- do your own research  :P

FrancescSobrevela, 2023-02-13:
Thanks for the response, and sorry for this issue.
I'm following the instructions as you said, but when I upload on the LilyGo
the compilation fails because a lack of libraries. These libraries
are: pca9555, L58Touch and epd_driver three times, how can I solve it?
Thanks

Missatge de MarcelvLaar ***@***.***> del dia dg., 12 de febr.
2023 a les 23:17: